unitreerobotics/unitree_mujoco #9: Support for high-level control in sim

RayYoh, 2024-05-27:
Hi, thanks for your work. Does current version just supports low-level control? I have tried the high-level control code in [unitree_sdk2_python, sportmode_test](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/example/high_level/sportmode_test.py), it occurs error.

wuyangning, 2024-05-29:
You are right. 
Unitree mujoco is mainly provided to users for sim to real verification of their control algorithms, so current version only provides basic support of: 
- sensor data (IMU, joint position, joint velocity).
- wireless controller (Simulate using an Xbox controller).
- motor control.

wuyangning, 2024-05-29:
> Hi, thanks for your work. Does current version just supports low-level control? I have tried the high-level control code in [unitree_sdk2_python, sportmode_test](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/example/high_level/sportmode_test.py), it occurs error.

You are right. 
Unitree mujoco is mainly provided to users for sim to real verification of their control algorithms, so current version only provides basic support of: 
- sensor data (IMU, joint position, joint velocity).
- wireless controller (Simulate using an Xbox controller).
- motor control.

closed 2024-12-11 by wuyangning