unitreerobotics/unitree_mujoco #19: cannot find -lddsc&ddscxx

Sher1ockFan, 2024-08-15:
my system is ubuntu20.04
I have installed the unitreeSDK2 on my computer
the path is /usr/local/
but there is an error about cannot find -lddsc&ddscxx occurred

juanjqo, 2024-08-21:
@Sher1ockFan I'm experiencing the same issue. Did you find a solution?

webDrag0n, 2024-08-23:
Same issue here, the -lddsc error could be solved by `apt install cyclonedds-dev`. However the -lddscxx still remains unfound. So I'm suspecting the problem is somehow related to building procedure of cyclonedds or the environment variable.

KanonWY, 2024-08-23:
I don't know if the maintainer of the repository made a deliberate mistake or is not familiar with modern cmake.The solutions to these errors are:

webDrag0n, 2024-08-23:
Thanks so much for the solution! It worked flawlessly.

Sher1ockFan, 2024-08-26:
> @Sher1ockFan I'm experiencing the same issue. Did you find a solution?

I found the installation path was wrong
It was /usr/local/lib/x86_64/ but it should be /usr/local/
So you can move libddsc.so and libddscxx.so
/ to the correct path
sudo mv /usr/local/lib/x86_64/* /usr/local/lib/

Sher1ockFan, 2024-08-26:
> @Sher1ockFan I'm experiencing the same issue. Did you find a solution?

I found the installation path was wrong
It was /usr/local/lib/x86_64/ but it should be /usr/local/lib
So you can move libddsc.so and libddscxx.so to the correct path
sudo mv /usr/local/lib/x86_64/* /usr/local/lib/

KanonWY, 2024-08-27:
It looks like a bug in the unitree_sdk2 cmake install script, there is a missing /, which causes the prefix. You can add this / and it will install to /usr/local.

liujieluv, 2024-08-31:
编译unitree_mujoco时出现，cannot find -lddsc&ddscxx #19 

#查找libddscxx.so 

locate libddscxx.so 

#将libddscxx.so的位置链接到/usr/local/lib/libddscxx.so 

sudo ln -s /home/liu/unitree_sdk2-main/thirdparty/lib/x86_64/libddscxx.so /usr/local/lib/libddscxx.so 

locate libddsc.so 

sudo ln -s /home/liu/unitree_sdk2-main/thirdparty/lib/x86_64/libddsc.so /usr/local/lib/libddsc.so

wuyangning, 2024-09-03:
> my system is ubuntu20.04 I have installed the unitreeSDK2 on my computer the path is /usr/local/ but there is an error about cannot find -lddsc&ddscxx occurred

The previous version of unitree_sdk2 was installed using install.sh. This method is no longer used in the new version. See https://github.com/unitreerobotics/unitree_sdk2. The installation paths of these two versions are inconsistent, resulting in compilation errors.
We have fixed this issue in the latest submission.