unitreerobotics/unitree_mujoco #22: MPR solver is removed from Mujoco, resulting c++ compiling failed

Revindell, 2024-09-05:
Hi, 

in the newest Mujoco release, the MPR solver is removed from mjOption (see [link](https://mujoco.readthedocs.io/en/latest/APIreference/APItypes.html#mjoption)), which will result in an error when compiling the C++ library.

[image]

toushei, 2024-09-07:
Try using the older version of [Mujoco 3.2.1](https://github.com/google-deepmind/mujoco/releases/tag/3.2.1). It seems like the developers used this version specifically since otherwise it gives library/header version mismatches.

KanonWY, 2024-09-08:
I made some modifications according to the error report, and it can run successfully. I hope it helps. maybe you can use Mujoco3.2.1 just like @toushei  des. [link](https://github.com/KanonWY/unitree_mujoco)

ZMcc18, 2024-10-16:
The newly installed mujuco version should be 3.2.4, and the error is found to say that there is no such member under the mjOpetion_, you can find the mjOption_ class in mjmodel.h under /usr/local/include/mujuco, and found that the ccd_iterations has been added, and mpr has been deleted, maybe you can try to unitree_mujoco/simulate/src/ Where the mujoco/simulation.cc error is reported, change the corresponding MPR to ccd

wuyangning, 2024-12-11:
We have modified the simulate program based on MuJoCo version 3.2.7 and fixed this problem. See: [f5b7837](https://github.com/unitreerobotics/unitree_mujoco/commit/f5b7837f8384dd50a15173fd67886bfdaef70f36)