unitreerobotics/unitree_mujoco #27: Unable to Control Robot Movement in Python Simulator

BeNavon, 2024-11-11:
Hello, I have successfully installed and tested the Python simulator, but I am currently unable to control the robot's movement in the simulation. While exploring further, I found that the examples in the "example" folder focus primarily on "Sim to Real" transitions and did not provide the guidance I need for controlling robot movement in the simulation.
Is there any additional documentation or more focused code examples available for controlling robot movement in the simulation environment?
Thank you for your help!

wuyangning, 2024-12-11:
High control and Client API are not support for this project.
This repository mainly used for sim to real verification of low level controller (MPC or RL).

DaySurprise, 2025-05-22:
我也成功启动了仿真环境，但是运行example/stand_go2.py示例文件的时候，mujoco仿真环境能够看到control参数在动态变化，但是机器狗go2没有任何动作，我想请问下怎么回事呢?