unitreerobotics/unitree_mujoco #29: Support high level control e.g. Walk/Get Balance Mode/... in Mujoco to control Unitree G1?

ChuangYan2, 2024-11-22:
Hello Unitree Team, 

Thanks for your awesome project. It helps us a lot. 

now we have a question, does the unitree_mujoco can serve as Client API to simulate the G1 high level control. 
I found the API from unitree_sdk2 can only work with real robot. Is that correct?

And do you have also suggestion for us, if we want to just simulate the high level control in mujoco without connection to real G1, is that possible? or how can we do it?

Thanks for any feedback

wuyangning, 2024-12-11:
High control and Client API are not support for this project. 
This repository mainly used for sim to real verification of low level controller (MPC or RL).

Rohith-Scalers, 2025-04-14:
any alternative for this ?

AntoniBertel, 2025-12-24:
Please let me know if you find anything; it’s confusing that it doesn’t come with locomotion