unitreerobotics/unitree_mujoco #31: compile problems

XX-4869, 2024-11-26:
xx@pc:~/unitree_mujoco/simulate/build$ make -j4
[ 23%] Built target jstest
[ 46%] Building CXX object CMakeFiles/unitree_mujoco.dir/src/mujoco/simulate.cc.o
[ 46%] Built target test
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc: In function ‘void {anonymous}::MakePhysicsSection(mujoco::Simulate*, int)’:
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:748:48: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_iterations’; did you mean ‘ls_iterations’?
  748 |         {mjITEM_EDITINT, "MPR Iter", 2, &(opt->mpr_iterations), "1 0 1000"},
      |                                                ^~~~~~~~~~~~~~
      |                                                ls_iterations
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:749:47: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_tolerance’; did you mean ‘ls_tolerance’?
  749 |         {mjITEM_EDITNUM, "MPR Tol", 2, &(opt->mpr_tolerance), "1 0 1"},
      |                                               ^~~~~~~~~~~~~
      |                                               ls_tolerance
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc: In member function ‘void mujoco::Simulate::Sync()’:
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2111:9: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_tolerance’; did you mean ‘ls_tolerance’?
 2111 |       X(mpr_tolerance);
      |         ^~~~~~~~~~~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2100:39: note: in definition of macro ‘X’
 2100 |   if (IsDifferent(scnstate_.model.opt.name, mjopt_prev_.name)) \
      |                                       ^~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2111:9: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_tolerance’; did you mean ‘ls_tolerance’?
 2111 |       X(mpr_tolerance);
      |         ^~~~~~~~~~~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2100:57: note: in definition of macro ‘X’
 2100 |   if (IsDifferent(scnstate_.model.opt.name, mjopt_prev_.name)) \
      |                                                         ^~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2111:9: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_tolerance’; did you mean ‘ls_tolerance’?
 2111 |       X(mpr_tolerance);
      |         ^~~~~~~~~~~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2103:18: note: in definition of macro ‘X’
 2103 |     Copy(m_->opt.name, scnstate_.model.opt.name);              \
      |                  ^~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2111:9: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_tolerance’; did you mean ‘ls_tolerance’?
 2111 |       X(mpr_tolerance);
      |         ^~~~~~~~~~~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2103:44: note: in definition of macro ‘X’
 2103 |     Copy(m_->opt.name, scnstate_.model.opt.name);              \
      |                                            ^~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2127:9: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_iterations’; did you mean ‘ls_iterations’?
 2127 |       X(mpr_iterations);
      |         ^~~~~~~~~~~~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2100:39: note: in definition of macro ‘X’
 2100 |   if (IsDifferent(scnstate_.model.opt.name, mjopt_prev_.name)) \
      |                                       ^~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2127:9: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_iterations’; did you mean ‘ls_iterations’?
 2127 |       X(mpr_iterations);
      |         ^~~~~~~~~~~~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2100:57: note: in definition of macro ‘X’
 2100 |   if (IsDifferent(scnstate_.model.opt.name, mjopt_prev_.name)) \
      |                                                         ^~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2127:9: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_iterations’; did you mean ‘ls_iterations’?
 2127 |       X(mpr_iterations);
      |         ^~~~~~~~~~~~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2103:18: note: in definition of macro ‘X’
 2103 |     Copy(m_->opt.name, scnstate_.model.opt.name);              \
      |                  ^~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2127:9: error: ‘mjOption’ {aka ‘struct mjOption_’} has no member named ‘mpr_iterations’; did you mean ‘ls_iterations’?
 2127 |       X(mpr_iterations);
      |         ^~~~~~~~~~~~~~
/home/xx/unitree_mujoco/simulate/src/mujoco/simulate.cc:2103:44: note: in definition of macro ‘X’
 2103 |     Copy(m_->opt.name, scnstate_.model.opt.name);              \
      |                                            ^~~~
make[2]: *** [CMakeFiles/unitree_mujoco.dir/build.make:115：CMakeFiles/unitree_mujoco.dir/src/mujoco/simulate.cc.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:134：CMakeFiles/unitree_mujoco.dir/all] 错误 2
make: *** [Makefile:84：all] 错误 2

XX-4869, 2024-11-27:
I change mpr_iterations and mpr_tolerance to ls_iterations and ls_tolerance, and then there's no problem, there's an error when I run ./unitree_mujoco 

xx@pc:~/unitree_mujoco/simulate/build$ ./unitree_mujoco 
MuJoCo version 3.2.6
Mujoco data is prepared
logConfigFileName:/home/xx/unitree_mujoco/simulate/build/default_logconf.json
1732678157.480751 [1] unitree_mu: selected interface "lo" is not multicast-capable: disabling multicast
Error: Joystick open failed.
段错误 (核心已转储)

Hotsburp, 2024-11-28:
See issue#11 [here](https://github.com/unitreerobotics/unitree_mujoco/issues/11#issuecomment-2452887941). Basically, set use_joystick as 0 in the unitree_mujoco/simulate/config.yaml instead of 1. Hope it helps.

jinsongyu-fb, 2024-12-05:
This is because mujoco renamed some fields. The solution is to pull tag 3.2.0 from mujoco and then build from source:

```
git clone https://github.com/google-deepmind/mujoco.git
cd mujoco
git checkout tags/3.2.0
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=~/local/mujoco-3.2.0
make -j32
make install
```

Then in this repro, `unitree_mujoco/simulate/CMakeList.txt`, add `list(APPEND CMAKE_PREFIX_PATH "~/local/mujoco-3.2.0")`  before running cmake

wuyangning, 2024-12-11:
We have modified the simulate program based on MuJoCo version 3.2.7 and fixed this problem.
See: https://github.com/unitreerobotics/unitree_mujoco/commit/f5b7837f8384dd50a15173fd67886bfdaef70f36

wuyangning, 2024-12-11:
> xx@pc:~/unitree_mujoco/simulate/build$ make -j4 [ 23%] Built target jstest [ 46%] Building CXX object CMakeFiles/unitree_mujoco.dir/src/mujoco/simulate.cc.o [ 46%] Built target test [...]

 We have modified the simulate program based on MuJoCo version 3.2.7 and fixed this problem. See: [f5b7837](https://github.com/unitreerobotics/unitree_mujoco/commit/f5b7837f8384dd50a15173fd67886bfdaef70f36)

Tipriest, 2025-06-20:
Now we had better use tag 3.2.3, because this repo has begun to use ccd_iterations in mjOption_, which occured from mujoco tag version 3.2.3.
the code should be

```bash
git clone https://github.com/google-deepmind/mujoco.git
cd mujoco
git checkout tags/3.2.3
mkdir build && cd build
cmake .. 
make -j32
sudo make install
```