unitreerobotics/unitree_mujoco #42: Request for example with G1 robot

Fatyhich, 2025-01-26:
Hello, dear developers ! 

I'm trying to work with G1 unitree robot in simulator. It successfully running and I can see robot on scene, but I can't control it :(

We have only 1 example for Go2 robot, and it working based on `unitree_go idl`. But for G1 we need to use `unitree_hg idl`

And there is critical difference, for example in command message:
from [pocket ](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/unitree_sdk2py/idl/unitree_go/msg/dds_/_LowCmd_.py) `unitree_go` for [class](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/unitree_sdk2py/idl/unitree_go/msg/dds_/_LowCmd_.py#:~:text=%E2%80%8E-,lowcmd_%E2%80%8E,-DefinitionSearch) `LowCmd_`:

```python
class LowCmd_(idl.IdlStruct, typename="unitree_go.msg.dds_.LowCmd_"):
    head: types.array[types.uint8, 2]
    level_flag: types.uint8
    frame_reserve: types.uint8
    sn: types.array[types.uint32, 2]
    version: types.array[types.uint32, 2]
    bandwidth: types.uint16
    motor_cmd: types.array['unitree_sdk2py.idl.unitree_go.msg.dds_.MotorCmd_', 20]
    bms_cmd: 'unitree_sdk2py.idl.unitree_go.msg.dds_.BmsCmd_'
    wireless_remote: types.array[types.uint8, 40]
    led: types.array[types.uint8, 12]
    fan: types.array[types.uint8, 2]
    gpio: types.uint8
    reserve: types.uint32
    crc: types.uint32
``` 

and same [class](https://github.com/unitreerobotics/unitree_sdk2_python/blob/3af1610228977f449e22c8fba868bb8516d41ec3/unitree_sdk2py/idl/unitree_hg/msg/dds_/_LowCmd_.py#L24-L29) `LowCmd_` but from [pocket](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/unitree_sdk2py/idl/unitree_hg/msg/dds_/_LowCmd_.py) `unitree_hg`:

```python
class LowCmd_(idl.IdlStruct, typename="unitree_hg.msg.dds_.LowCmd_"):
    mode_pr: types.uint8
    mode_machine: types.uint8
    motor_cmd: types.array['unitree_sdk2py.idl.unitree_hg.msg.dds_.MotorCmd_', 35]
    reserve: types.array[types.uint32, 4]
    crc: types.uint32
```

They are so different and for me hard to understand how to adopt example with Go2 for G1 at least to control robot.

In example we have [code](https://github.com/unitreerobotics/unitree_mujoco/blob/f5b7837f8384dd50a15173fd67886bfdaef70f36/example/python/stand_go2.py#L40-L51) which is configuring msg for motors. how to deal with new hg version ? any suggestions or source where I can read about it ?

jloganolson, 2025-03-28:
Hope this helps: https://logansrobot.bearblog.dev/g1-mujoco-python-getting-started/

kitare17, 2025-09-04:
> Hope this helps: https://logansrobot.bearblog.dev/g1-mujoco-python-getting-started/

thank you

Agnel-Wang, 2025-09-28:
Thanks to jlganolson.

And compared to unitree_sdk2py, I have a personal simplified [repository](https://github.com/Agnel-Wang/unitree_dds_wrapper) that might also help you.