unitreerobotics/unitree_mujoco #46: Crash on running `./unitree_mujoco`: free(): invalid pointer

BeNavon, 2025-02-27:
When I run ./unitree_mujoco using MuJoCo version 3.2.7 (the tested version), the application prints the following messages and then crashes:
```
MuJoCo version 3.2.7
Mujoco data is prepared
free(): invalid pointer
Aborted
```
Detailed backtrace:
```
Thread 2 "unitree_mujoco" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffef600640 (LWP 1436694)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737209435712) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
```

Environment:
- Unitree_mujoco: f5b7837
- MuJoCo version: 3.2.7
- DDS Library: Cyclone DDS 0.10.5-2jammy.20241127.232046
- RMW Implementation: ros-humble-rmw-cyclonedds-cpp 1.3.4-1jammy.20241128.004150
- OS: Ubuntu 22.04

Steps to Reproduce:
1. Build unitree_mujoco according to the provided instructions (using MuJoCo 3.2.7).
2. Run `./unitree_mujoco` in the build directory.
3. Observe the crash with the error message `free(): invalid pointer`.

Could you please advise on how to resolve this issue or if any additional configuration changes are needed for compatibility?

BeNavon, 2025-03-12:
> 请问解决了么 我也遇到了同样的问题

"I have the same problem. Have you solved it?" 
No, and I haven’t received any response from Unitree support.

JWishere, 2025-03-18:
did you try sudo ./unitree_mujoco ? It works for me

yitingHH-bit, 2025-04-04:
> did you try sudo ./unitree_mujoco ? It works for me

doesnot work..

sudo ./unitree_mujoco
./unitree_mujoco: error while loading shared libraries: libi

Jiachenggg, 2025-04-08:
> did you try sudo ./unitree_mujoco ? It works for me
Thanks a lot! I work

zitongbai, 2025-08-15:
It seems that if you have ros2 humble installed, some confict in dds might cause the problem. I solved this by changing the dds implementation according to the [unitree official doc](https://support.unitree.com/home/en/G1_developer/ros2_communication_routine). 
 
For ros2 humble:

First install the dependences:
```bash
sudo apt install ros-humble-rmw-cyclonedds-cpp
sudo apt install ros-humble-rosidl-generator-dds-idl
```

Before compiling cyclonedds, please ensure that ros2 environment has NOT been sourced when starting the terminal. Otherwise, it may cause errors in compilation.

If `source/opt/ros/humble/setup.bash` has been added to the `~/.bashrc` file when installing ROS2, it needs to be commented out:

```bash
sudo apt install gedit
sudo gedit ~/.bashrc
```

```bash
# in .bashrc
# source /opt/ros/humble/setup.bash 
```

Then build cyclone-dds:

```bash
mkdir -p ~/cyclonedds_ws/src
cd ~/cyclonedds_ws/src
git clone https://github.com/ros2/rmw_cyclonedds -b humble
git clone https://github.com/eclipse-cyclonedds/cyclonedds -b releases/0.10.x 
cd ..
colcon build --packages-select cyclonedds #Compile cyclone-dds package
source /opt/ros/humble/setup.bash # source ROS2 environment
colcon build # Compile all packages in the workspace
```

After that, you can rewrite the `source/opt/ros/humble/setup.bash` in `.bashrc`:
```bash
# in .bashrc
source /opt/ros/humble/setup.bash
source ~/cyclonedds_ws/install/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

What's more, unitree_sdk2 use DOMAIN_ID 0 in DDS, so sometimes we need to set the `ROS_DOMAIN_ID` to some other number except 0. 
```bash
# Optional
export ROS_DOMAIN_ID=10 # or other number you like
```

Agnel-Wang, 2025-10-14:
Do not mix unitree_sdk2’s dds.so with ROS’s dds.so.
Use ldd to check which shared objects are currently linked and to verify the effective LD_LIBRARY_PATH.

closed 2025-10-14 by Agnel-Wang

vegewenqi, 2026-03-22:
> did you try sudo ./unitree_mujoco ? It works for me

works for me as well with ”sudo"