unitreerobotics/unitree_mujoco #47: Joystick does not move Go2

tamutamuto, 2025-03-02:
## **Overview**  
I read the README as saying that setting USE_JOYSTICK = 1 allows controlling Go2 with a joystick.
However, in reality, the joystick input was not reflected in the movement of the robot..

To confirm this issue, I added a debug `print()` statement in `unitree_sdk2py_bridge.py` and verified that joystick input is correctly being received.

##  Actual Behavior
### **🔹 Steps to Reproduce**
1. Set `USE_JOYSTICK = 1` in `simulate_python/config.py`
2. Run `python3 unitree_mujoco.py`
3. Operate the joystick
4. Add a debug `print()` statement in `PublishWirelessController()` in `unitree_sdk2py_bridge.py`
5. Confirm that joystick input is received but the robot does not move at all

##  Added Debug Code**
I added the following debug `print()` statement in `PublishWirelessController()` in `unitree_sdk2py_bridge.py`:

```python
def PublishWirelessController(self):
    if self.joystick is not None:
        pygame.event.get()
        
        key_state = [0] * 16
        key_state[self.key_map["R1"]] = self.joystick.get_button(self.button_id["RB"])
        key_state[self.key_map["L1"]] = self.joystick.get_button(self.button_id["LB"])
        key_state[self.key_map["start"]] = self.joystick.get_button(self.button_id["START"])
        key_state[self.key_map["select"]] = self.joystick.get_button(self.button_id["SELECT"])
        key_state[self.key_map["R2"]] = (self.joystick.get_axis(self.axis_id["RT"]) > 0)
        key_state[self.key_map["L2"]] = (self.joystick.get_axis(self.axis_id["LT"]) > 0)
        key_state[self.key_map["F1"]] = 0
        key_state[self.key_map["F2"]] = 0
        key_state[self.key_map["A"]] = self.joystick.get_button(self.button_id["A"])
        key_state[self.key_map["B"]] = self.joystick.get_button(self.button_id["B"])
        key_state[self.key_map["X"]] = self.joystick.get_button(self.button_id["X"])
        key_state[self.key_map["Y"]] = self.joystick.get_button(self.button_id["Y"])
        key_state[self.key_map["up"]] = self.joystick.get_hat(0)[1] > 0
        key_state[self.key_map["right"]] = self.joystick.get_hat(0)[0] > 0
        key_state[self.key_map["down"]] = self.joystick.get_hat(0)[1] < 0
        key_state[self.key_map["left"]] = self.joystick.get_hat(0)[0] < 0

        key_value = 0
        for i in range(16):
            key_value += key_state[i] << i

        self.wireless_controller.keys = key_value
        self.wireless_controller.lx = self.joystick.get_axis(self.axis_id["LX"])
        self.wireless_controller.ly = -self.joystick.get_axis(self.axis_id["LY"])
        self.wireless_controller.rx = self.joystick.get_axis(self.axis_id["RX"])
        self.wireless_controller.ry = -self.joystick.get_axis(self.axis_id["RY"])

        # Output joystick state here
        print(f"Joystick State:")
        print(f"  LX: {self.wireless_controller.lx:.2f}, LY: {self.wireless_controller.ly:.2f}")
        print(f"  RX: {self.wireless_controller.rx:.2f}, RY: {self.wireless_controller.ry:.2f}")
        print(f"  Keys: {bin(self.wireless_controller.keys)}")

        self.wireless_controller_puber.Write(self.wireless_controller)
```

##  Actual Output (When Moving Joystick)
After running this code, I confirmed that joystick input is correctly received when I moved the joystick, as shown below:

```yaml
Joystick State:
  LX: 0.00, LY: 1.00
  RX: 0.00, RY: 0.00
  Keys: 0b0
Joystick State:
  LX: 0.00, LY: 1.00
  RX: 0.00, RY: 0.00
  Keys: 0b0
Joystick State:
  LX: 0.00, LY: 1.00
  RX: 0.00, RY: 0.00
  Keys: 0b0
Joystick State:
  LX: 0.00, LY: 1.00
  RX: 0.00, RY: 0.00
  Keys: 0b0
Joystick State:
  LX: 0.00, LY: 1.00
  RX: 0.00, RY: 0.00
  Keys: 0b0
Joystick State:
  LX: 0.00, LY: 1.00
  RX: 0.00, RY: 0.00
  Keys: 0b0
```
Despite receiving joystick input, the robot does not move.

##  Issue 
PublishWirelessController() only sends data to rt/wireless_controller and does not affect the LowCmd controlling the robot.
I think the README misleads users into thinking that enabling the joystick allows direct control of the robot.
I also couldn't find any reports of anyone successfully moving Go2 using the joystick.

## Request
The README is misleading and should be corrected to clarify that enabling `USE_JOYSTICK = 1` does not allow direct control of the robot.  
If joystick support for robot movement is planned, it would be helpful to update the documentation accordingly.

Bailey-24, 2025-03-26:
yes, I meet the same issues

andy-zhuo-02, 2025-04-01:
Moreover, in the C++ version,  output of exec 'jstest' is also not friendly to the one who encounter this project for the first time.
`cout << unitree_key.value << endl;` always output a value that does not change. I guess the maintainer wants us to modify the value to what we want, but it should leave a msg here.

Agnel-Wang, 2025-09-28:
use_joystick = 1 does not actually allow you to directly control the robot with a remote controller inside unitree_mujoco.
It only passes the joystick information out.
The actual use of the joystick happens in other control programs, such as the G1 control program in unitree_rl_lab, which uses the joystick information from LowState as input mapped to control commands.

closed 2025-09-28 by Agnel-Wang