unitreerobotics/unitree_mujoco #48: error: \u2018Eigen::QuaternionBase<Eigen::Quaternion<float> >::Matrix3 {aka \u2018class Eigen::Matrix<float, 3, 3>\u2019} has no member named canonicalEulerAngles\u2019

kExU9853, 2025-03-05:
unitree_mujoco/simulate/src/unitree_sdk2_bridge/unitree_sdk2_bridge.cc:97:59: error: \u2018Eigen::QuaternionBase<Eigen::Quaternion<float> >::Matrix3\u2019 {aka \u2018class Eigen::Matrix<float, 3, 3>\u2019} has no member named \u2018canonicalEulerAngles\u2019
   97 |             Eigen::Vector3f ypr = quat.toRotationMatrix().canonicalEulerAngles(2, 1, 0);
      |                                                           ^~~~~~~~~~~~~~~~~~~~
/home/xk/unitree_mujoco/simulate/src/unitree_sdk2_bridge/unitree_sdk2_bridge.cc: In member function \u2018void UnitreeSdk2Bridge::PublishLowStateHg()\u2019:
/home/xk/unitree_mujoco/simulate/src/unitree_sdk2_bridge/unitree_sdk2_bridge.cc:146:59: error: \u2018Eigen::QuaternionBase<Eigen::Quaternion<float> >::Matrix3\u2019 {aka \u2018class Eigen::Matrix<float, 3, 3>\u2019} has no member named \u2018canonicalEulerAngles\u2019
  146 |             Eigen::Vector3f ypr = quat.toRotationMatrix().canonicalEulerAngles(2, 1, 0);
      |                                                           ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/unitree_mujoco.dir/build.make:146: CMakeFiles/unitree_mujoco.dir/src/unitree_sdk2_bridge/unitree_sdk2_bridge.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/unitree_mujoco.dir/all] Error 2

A1ic3-g, 2025-03-06:
Also have this error, seems to have been introduced in one of the commits by @Bin-Ro in the last few days.

Yandong-Luo, 2025-03-08:
I use `Eigen::Vector3f ypr = quat.toRotationMatrix().eulerAngles(2, 1, 0);` to replace the `Eigen::Vector3f ypr = quat.toRotationMatrix().canonicalEulerAngles(2, 1, 0);` And finish compile

kvbogdanov, 2025-03-10:
Made the changes suggested by @Yandong-Luo but got an error:

> root@sfuai-B760M-DS3H-DDR4:/usr/local/etc/unitree_mujoco/simulate# make -j4
> Consolidate compiler generated dependencies of target unitree_mujoco
> [ 23%] Built target jstest
> [ 30%] Linking CXX executable test
> /usr/bin/ld: cannot open output file test: Is a directory
> collect2: error: ld returned 1 exit status
> make[2]: *** [CMakeFiles/test.dir/build.make:100: test] Error 1
> make[1]: *** [CMakeFiles/Makefile2:113: CMakeFiles/test.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> [ 38%] Building CXX object CMakeFiles/unitree_mujoco.dir/src/unitree_sdk2_bridge/unitree_sdk2_bridge.cc.o
> [ 46%] Linking CXX executable unitree_mujoco
> [ 92%] Built target unitree_mujoco
> make: *** [Makefile:91: all] Error 2

Bin-Ro, 2025-03-12:
I have already fixed the bug, I don't use eigen anymore

Bin-Ro, 2025-03-12:
I’m so sorry for making so many mistakes, but I have already reverted them to the original. You can use the latest commit.