unitreerobotics/unitree_mujoco #51: 运行./unitree_mujoco 报错

CTGUMARK, 2025-03-11:
采取配置文件:
robot: "go2"  # Robot name, "go2", "b2", "b2w", "h1", "go2w", "g1"
robot_scene: "scene.xml" # Robot scene, /unitree_robots/[robot]/scene.xml 

domain_id: 0  # Domain id
interface: "enp0s31f6" # Interface 

use_joystick: 0 # Simulate Unitree WirelessController using a gamepad 
joystick_type: "xbox" # support "xbox" and "switch" gamepad layout
joystick_device: "/dev/input/js0" # Device path
joystick_bits: 16 # Some game controllers may only have 8-bit accuracy

print_scene_information: 1 # Print link, joint and sensors information of robot

enable_elastic_band: 0 # Virtual spring band, used for lifting h1

但是运行报错:

root@wu-System-Product-Name:/home/wu/slam_ws/mujoco_ws/src/unitree_mujoco/simulate/build# ./unitree_mujoco 
MuJoCo version 3.3.1
Mujoco data is prepared
Error: Joystick open failed.

Yandong-Luo, 2025-03-11:
I also meet this situation. But I change to 2f5459d466ba72b14b97fc15d215fa0516bcc19f. It can avoid strong dependence on joystick。

Bin-Ro, 2025-03-12:
I’m so sorry for making so many mistakes, but I have already reverted them to the original. You can use the latest commit.

CTGUMARK, 2025-03-12:
更新版本之后,但是在运行过程中同样遇到: 
root@wu-System-Product-Name:/home/wu/slam_ws/mujoco_ws/src/unitree_mujoco/simulate/build# ./unitree_mujoco 
MuJoCo version 3.3.1
Mujoco data is prepared
free(): invalid pointer
Aborted (core dumped)

lexand59, 2025-03-19:
I understand that the problem has not yet been resolved and that unitree_mujoco is not currently supported on Ubuntu 22.04. Is this correct?

wmh02240, 2025-04-12:
sudo ./unitree_mujoco

canwisher, 2025-08-25:
> mujoco版本是3.2.7时，遇到相同问题: 
cdw@cdw-computer:~/unitree/unitree_mujoco/simulate/build$ ./unitree_mujoco 
MuJoCo version 3.2.7
Mujoco data is prepared
1756118845.399939 [1] unitree_mu: selected interface "lo" is not multicast-capable: disabling multicast
free(): invalid pointer
已中止 (核心已转储)

canwisher, 2025-08-25:
> > 更新版本之后,但是在运行过程中同样遇到: root@wu-System-Product-Name:/home/wu/slam_ws/mujoco_ws/src/unitree_mujoco/simulate/build# ./unitree_mujoco MuJoCo version 3.3.1 Mujoco data is prepared free(): invalid pointer Aborted (core dumped)
> 
> mujoco版本是3.2.7时，遇到相同问题: cdw@cdw-computer:~/unitree/unitree_mujoco/simulate/build$ ./unitree_mujoco MuJoCo version 3.2.7 Mujoco data is prepared 1756118845.399939 [1] unitree_mu: selected interface "lo" is not multicast-capable: disabling multicast free(): invalid pointer 已中止 (核心已转储)

我已解决 free(): invalid pointer 问题：

- 操作系统：ubuntu24.04
- Ros版本：Ros2 Jazzy

通过命令`ldd ./unitree_mujoco | grep dds`发现./unitree_mujoco的dds包路径发生冲突，混合使用了jazzy自带的dds包的路径和unitree_robotics中通过[unitree_sdk2](https://github.com/unitreerobotics/unitree_sdk2)安装的dds包，导致 free(): invalid pointer 。

[image]<br/>

我通过配置 ~/.bashrc ，优先使用 Unitree SDK 自带的 DDS 库，避免和 ROS2 Jazzy 冲突：

- 在你的用户目录下编辑 ~/.bashrc
```
nano ~/.bashrc
``` 

- 在你的bashrc文件末尾加上
```
# >>> Unitree DDS setup >>>
export UNITREE_DDS_PATH=/opt/unitree_robotics/lib
export LD_LIBRARY_PATH=$UNITREE_DDS_PATH:$LD_LIBRARY_PATH
# <<< Unitree DDS setup <<<
``` 

- 保存退出，然后执行
```
source ~/.bashrc
``` 

- 最终，重新编译unitree_mujoco，让配置生效。

closed 2025-09-26 by Agnel-Wang