unitreerobotics/unitree_mujoco #53: unitree_sdk2_python installation

mohamed-djeha-rob, 2025-03-23:
When installing unitree_sdk2_python, I run into the already documented issue:
`Could not locate cyclonedds. Try to set CYCLONEDDS_HOME or CMAKE_PREFIX_PATH`

I followed exactly the steps to fix this issue. However, I still have the same error message when attempting to install unitree_sdk2_python.

ddf1826120803, 2025-04-08:
Also encountered this issue, you can try ```export CYCLONEDDS_HOME="/home/<user>/cyclonedds/install"``` instead of ```export CYCLONEDDS_HOME="~/cyclonedds/install"```.

mohamed-djeha-rob, 2025-04-11:
I tried this solution before posting the issue, and unfortunately it did not work...

Jerryguo778, 2025-08-29:
> I tried this solution before posting the issue, and unfortunately it did not work...

After switching to the root user, execute the `export` command, and then try executing `pip`. I encountered the same issue in my environment, which was resolved by switching to the root user