unitreerobotics/unitree_mujoco #61: Interaction with the unitree_ros2 problem

kvbogdanov, 2025-04-25:
Ubuntu 22 + ROS2 

I've started simulation by a command  _sudo ./unitree_mujoco_ it started without any errors but nothing happens further (robot falls down) 

On another terminal (in the unitree_ros2 folder) I did _source ./setup_local.sh_ and after that _ros topic list_ shows only:
/parameter_events
/rosout

I've also got xbox joystic (tested and works fine), but robot ignore it either

What could be wrong with this configuration?

Agnel-Wang, 2025-05-08:
The default dds domain id in simulte/config.yaml is 1. (which is used for avoiding conflicts between simulation and real robot)
You should set it to zero, or set ros domain id by
export ROS_DOMAIN_ID=1

kvbogdanov, 2025-05-16:
@Agnel-Wang thanks! 

Now I have a longer list of topics, but it's still incomplete. The Issues section in unitree_ros2 is closed, but I suspect the problem lies in that part. I encountered an error when trying to execute 'start_sport_mode_ctrl' or 'read_wireless_controller':
`error while loading shared libraries: libunitree_go__roidl_typesupport_cpp.so`

[image]

How can I deal with it?