unitreerobotics/unitree_mujoco #76: Compile Error unitree_mujoco

okadahiroyuki, 2025-07-27:
cd unitree_mujoco/simulate
mkdir build && cd build
cmake ..
make -j4

The following error appears

/home/roboworks/Mujoco/unitree_mujoco/simulate/src/mujoco/simulate.cc:3041:62: error: ‘mjv_updateSceneFromState’ was not declared in this scope
 3041 |           scnstate_.data.warning[mjWARN_VGEOMFULL].number += mjv_updateSceneFromState(
      |                                                              ^~~~~~~~~~~~~~~~~~~~~~~~
/home/roboworks/Mujoco/unitree_mujoco/simulate/src/mujoco/simulate.cc:3086:27: error: ‘scnstate_’ was not declared in this scope; did you mean ‘setstate_r’?
 3086 |       mjv_freeSceneState(&scnstate_);
      |                           ^~~~~~~~~
      |                           setstate_r
/home/roboworks/Mujoco/unitree_mujoco/simulate/src/mujoco/simulate.cc:3086:7: error: ‘mjv_freeSceneState’ was not declared in this scope; did you mean ‘mjv_freeScene’?
 3086 |       mjv_freeSceneState(&scnstate_);
      |       ^~~~~~~~~~~~~~~~~~
      |       mjv_freeScene
make[2]: *** [CMakeFiles/unitree_mujoco.dir/build.make:132: CMakeFiles/unitree_mujoco.dir/src/mujoco/simulate.cc.o] エラー 1
make[2]: *** 未完了のジョブを待っています....
make[2]: *** [CMakeFiles/unitree_mujoco.dir/build.make:160: CMakeFiles/unitree_mujoco.dir/src/main.cc.o] エラー 1
[ 92%] Linking CXX executable test
[ 92%] Built target test
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/unitree_mujoco.dir/all] エラー 2
make: *** [Makefile:91: all] エラー 2

weihaosky, 2025-07-28:
I met the same problem. Anyone can help? Many thanks!

weihaosky, 2025-07-28:
> I met the same problem. Anyone can help? Many thanks!

In my case it is due to the mujoco version. Using mujoco 3.2.7 resolves this.

katsully, 2025-08-17:
I'm using mujoco 3.2.7 and still having the same error!

Hotsburp, 2025-09-02:
@katsully 

There may have been some issues with the newest version of unitree_mujoco. 

To play safe, install mujoco 3.2.7 and install unitree_sdk2 (commit [45cac25](https://github.com/unitreerobotics/unitree_sdk2/tree/45cac25a5b68f440b3439ffbdeb0dff7c1b053ec) into opt/unitree_robotics as per what @soowonic @weihaosky mentioned. For unitree mujoco, i installed the commit [2f5459d](https://github.com/unitreerobotics/unitree_mujoco/tree/2f5459d466ba72b14b97fc15d215fa0516bcc19f). 

This configuration worked for me, hope it helps!